kellerkindt/onewire #4: Focus library on generic one wire support

TDHolmes, 2019-09-01:
Hello, I'm looking to start my own crate specifically for a different one wire temperature sensor and stumbled upon this crate. It looks like there is good, generic one wire support in this crate, but support for the ds18b20 is mixed in here too. I think it makes sense to clean up this crate to focus on generic one wire support and perhaps spin ds18b20 out into its own crate that uses this crate. Let me know what you think.

riaqn, 2019-10-23:
An alternative approach would be to add support for other devices directly into this library.

TDHolmes, 2019-10-27:
True, but doesn't seem very scaleable. There's a lot of one wire chips out there

riaqn, 2019-10-28:
Yes, but I don't see how is that not scaleable.

Each chip will take like 2kb of source code. It won't be a big trouble.

I suggest you start by adding new devices to this library and create PR. Only consider spliting crates when it's clearly motivated.

closed 2019-12-30 by TDHolmes