kellerkindt/onewire #7: Cannot get this crate to work with atsamd-hal

TDHolmes, 2019-09-15:
Hey, I've been trying for the last few days to get this crate to work with my setup on an `atsamd` chip (specifically, the [`feather_m0`](https://github.com/atsamd-rs/atsamd/tree/master/boards/feather_m0) board), but have been unable to get it to work. It seems I can't get a pin that clearly implements both the input and output trait as required by the `OpenDrainOutput` trait.

Here's a slimmed down version of my code:

```rust
#![no_std]
#![no_main]

extern crate panic_halt;  // panic handler

use onewire;

use feather_m0 as hal;
use hal::prelude::*;
use hal::{entry, Peripherals, CorePeripherals};

#[entry]
fn main() -> ! {
    // cortex_m::Peripherals::take().unwrap()
    let mut peripherals = Peripherals::take().unwrap();

    let mut pins = hal::Pins::new(peripherals.PORT);
    let mut ow_pin = pins.d11.into_open_drain_output(&mut pins.port);
    let mut one_wire = onewire::OneWire::new(&mut ow_pin, false);
    loop {}
}
```

I get the following error:

```rust
error[E0277]: the trait bound `atsamd_hal::gpio::Pa16<atsamd_hal::gpio::Output<atsamd_hal::gpio::OpenDrain>>: embedded_hal::digital::v1::InputPin` is not satisfied
  --> src/main.rs:71:46
   |
71 |     let mut one_wire = onewire::OneWire::new(&mut ow_pin, false);
   |                                              ^^^^^^^^^^^ the trait `embedded_hal::digital::v1::InputPin` is not implemented for `atsamd_hal::gpio::Pa16<atsamd_hal::gpio::Output<atsamd_hal::gpio::OpenDrain>>`
   |
   = note: required because of the requirements on the impl of `embedded_hal::digital::v2::InputPin` for `atsamd_hal::gpio::Pa16<atsamd_hal::gpio::Output<atsamd_hal::gpio::OpenDrain>>`
   = note: required for the cast to the object type `dyn onewire::OpenDrainOutput<Error = (), Error = ()>`
```

In your example, you call `downgrade()` after converting to an open drain output, but it seems the `atsamd-hal` does not have this method on their GPIO HAL. Any ideas?

david-sawatzke, 2020-01-22:
The issue is that atsamd-hal doesn't implement `InputPin` on OpenDrain pins, only on Input pins (https://github.com/atsamd-rs/atsamd/blob/1badb46942c2ec6eed9a4f5fe498c0d220b6f811/hal/src/common/gpio.rs#L343). The onewire implementation needs this and the stm32f1xx-hal does provide it (https://github.com/stm32-rs/stm32f1xx-hal/blob/8b5251431af453d4075b52d1c4dcd456104243f7/src/gpio.rs#L303). Your best bet is to fix it (implement InputPin for OpenDrain pins) in the atsamd-hal or open an issue over there.

David-OConnor, 2020-05-23:
Same issue with `stm32f3xx_hal`, and `rppal`. I'm wondering if something else is going on, since they both seem to be `V2` compatible. Related question: On what HAL implementations does this crate work on, without triggering this error?

David-OConnor, 2020-06-02:
I patched the above crate to make it work. Was a quick fix.

David-OConnor, 2020-06-22:
It looks like the latest set of commits fixes it for me on f3.

TDHolmes, 2021-12-17:
This works with atsamd GPIO revamp

closed 2021-12-17 by TDHolmes