kellerkindt/onewire #13: Allow setting resolution?

riaqn, 2020-07-16:
Currently the resolution field is private.

Feriixu, 2025-02-28:
Just setting the field doesn't do anything. A command needs to be sent to the sensor, but that just isn't implemented at the moment.

kellerkindt, 2025-03-01:
PRs are always welcome. As described by @Feriixu, setting the [resolution attribute](https://github.com/kellerkindt/onewire/blob/master/src/ds18b20.rs#L46) for the DS18B20 alone is not sufficient. The non-default value must also be transmitted to the chip before requesting new reads. At the moment, no one has deemed this important enough to implement it.

riaqn, 2025-03-01:
Thank you for replying. I no longer use the library so will close the issue.

closed 2025-03-01 by riaqn