kellerkindt/onewire #14: always CRC mismatch unless parasite mode even the VCC is connected

riaqn, 2020-07-19:
Pretty much what the title says. With very high probability I get Crc mismatch error. If I set parasite mode=true when initializing, it works perfectly - but the VCC of the sensors are connected to 3.3V and property 4.7k ohm resistor.

kellerkindt, 2025-03-01:
Hi there. I am sorry, but it is hard to remotely debug your issue, especially as it most like seems related to your hardware setup. Could you provide more context? I am sure there will be people to help you. If you were able to solve the issue: even better! Could you then also share the solution?

riaqn, 2025-03-01:
Thank you getting back (after 5 years!). I no longer play with the devices so will mark the issue as resolved.