kellerkindt/onewire #17: Can't get this crate to work with embassy

maxhambraeus, 2022-12-08:
This is the code: (pretty much the readme example, modified slightly for embassy)
```
#![no_std]
#![no_main]
#![feature(type_alias_impl_trait)]

use embassy_executor::Spawner;
use embassy_rp::gpio::{Input, Level, OutputOpenDrain, Pull, Flex};
use {defmt_rtt as _, panic_probe as _};
use defmt::*;
use onewire::ds18b20::DS18B20;
use embedded_hal::blocking::delay::DelayMs;

#[embassy_executor::main]
async fn main(_spawner: Spawner) {
    let p = embassy_rp::init(Default::default());
    let mut temp_pin = OutputOpenDrain::new(p.PIN_0, Level::High);

    let mut delay = embassy_time::Delay;
    
    let mut wire = onewire::OneWire::new(temp_pin, false);

    if wire.reset(&mut delay).is_err() {
        //missing pullup or error on line
        info!("nope");
        loop {}

    }

    let mut search = onewire::DeviceSearch::new();

    let a = wire.search_next(&mut search, &mut delay).unwrap();
    dbg!("{:?}", a.unwrap().address);

    while let Some(device) = wire.search_next(&mut search, &mut delay).unwrap() {
        info!("got here!");
        match device.address[0] {
            onewire::ds18b20::FAMILY_CODE => {
                let mut ds18b20 = DS18B20::new(device).unwrap();

                //request sensor to measure temperature
                let resolution = ds18b20.measure_temperature(&mut wire, &mut delay).unwrap();

                delay.delay_ms(resolution.time_ms());

                let temperature = ds18b20.read_temperature(&mut wire, &mut delay).unwrap();
                println!("{}", temperature);
            },
            _ => {

            }
        }
    }

}
```
It compiles fine with cargo version 1.67.0 nightly, but when it is ran on a raspberry pi pico W it panics at line 30,  because the `wire.search_next` is `None`. I think my wiring is correct, as if I remove the pullup resistor (10k ohm) it complains about a missing pullup.

maxhambraeus, 2022-12-12:
Update: works fine with an Arduino, so neither my wiring or the sensor itself is at fault here.

otoshi1205, 2023-07-03:
Hello, is there any update on that?

Today I checked it on my setup. The code is exactly the same as @maxhambraeus's one. Works with Arduino, but not with the embassy. I plan to check with an oscilloscope tomorrow, and hopefully, I'll have more information about it.

Feriixu, 2025-02-28:
@otoshi1205 @maxhambraeus I added an example to this crate in another PR. Check it out!

@kellerkindt I think this can be closed.

kellerkindt, 2025-03-01:
See #22 . Feel free to open a new issue if the problem still persists.

closed 2025-03-01 by kellerkindt