kellerkindt/onewire #19: Mention that you need certain optimization level

Huballs, 2024-09-27:
I'm new to rust so i just spent two days trying to make this work, this the only configuration that worked for me:

[profile.dev]
codegen-units = 2
debug = 0
lto = true
opt-level = 2

mcu - STM32L011F4Px, hal - stm32l0xx-hal

nebogeo, 2025-02-10:
I also spent a number of days trying to make this work (on a rp2040/pico board) until stumbling across this comment - so reporting in that this is a platform independent thing, and very difficult to spot...

kellerkindt, 2025-03-01:
We now have an example for the rp2024: see #22 . I think the need for certain optimization levels is kind of common for a lot of embedded projects. The new example should demonstrate that better now, with the Cargo.toml entry.

Feel free to open a new issue if the problem still persists.

closed 2025-03-01 by kellerkindt